adafruit/Adafruit_WICED_Arduino #34: Please compare your core (libmaple) with stm32duino (no issue)

madias123, 2016-03-29:
Just a friendly hint:
Please compare your used core (libmaple) with the one from stm32duino:
https://github.com/rogerclarkmelbourne/Arduino_STM32
It's also based on the leaflabs core and the whole community spent much work (and time) in the past two years to developed this  abandoned project further.

regards 
Matthias

ladyada, 2016-03-29:
Yes thank you - its a great core and a great library, we added F2 support too :)

closed 2016-03-29 by ladyada

madias123, 2016-03-29:
Keep in mind, as leaflabs started to do stm32 for *duino IDE the they have to build everything new, because the was no open source code for STM32. Meanwhile (after leaflabs abandoned the maple project) things are different: HAL and cubemx is open source, so if adafruit is planning to do more STM32 stuff it's maybe a good idea to observe the possibility to integrate the HAL/cubeMX core instead of the "old" libmaple one. I know for a single project this would be too much work. Some cores from our members are rudimentary based on HAL (STM32F3/F4). A good place to read everything about STM32 and Arduino integration is our forum at: http://stm32duino.com/
We have done many conversations of libraries (yes, even Adafruit ones :) !) to get working with STM32. Feel free to dig in!

ladyada, 2016-03-29:
OK we'll take a look! thanks for the note :)