adafruit/Adafruit_WICED_Arduino #35: HttpsLargeData.ino fails

bprobbins, 2016-04-01:
HttpsLargeData.ino fails as is ( invalid certificates error).
Requires addition of:
```
#include "certificates.h"
#define INCLUDE_DEFAULT_ROOTCA 0
```
as well as
`Feather.useDefaultRootCA(INCLUDE_DEFAULT_ROOTCA);
`
in setup()

microbuilder, 2016-04-01:
The AWS certificate is included by default on the WICED platform so 'certificates.h' isn't required and you shouldn't disable the default root CA list.

bprobbins, 2016-04-01:
Thanks, OK,  but I wonder why I repeatedly received invalid certificate message upon running the ino?

microbuilder, 2016-04-01:
Oh, sorry ... you're right ... something must have very recently changed with this.  Thanks for the heads up, I'll look into it and get back to you here:

```
HTTPS Large Data Example

Bootloader  : 1.0.0
WICED SDK   : 3.5.2
FeatherLib  : 0.5.0
Arduino API : 0.5.0

Please wait while connecting to: 'HOMENETWORK' ... Connected!

SSID        : HOMENETWORK (-88 dBm)
Encryption  : WPA2_AES
MAC Address : 44:39:C4:EB:BE:94
Local IP    : 10.0.1.30
Gateway     : 10.0.1.1
Subnet Mask : 255.255.255.0

Connecting to 'adafruit-download.s3.amazonaws.com' port 443 ... 
SDEP_CMD_TCP_CONNECT failed, Error: UNTRUSTED_CERTIFICATE (5035)

--- FEATHER HALTED ---
```

microbuilder, 2016-04-02:
It looks like AWS just switched from 'Digicert' to 'Baltimore' for their root CA.  We'll update the default root CA list to change these in the next FeatherLib update, and update this example ... thanks for bringing this to our attention!  I'll close this issue when the update is implemented.

microbuilder, 2016-04-05:
I've updated this example in the development branch to include the 'baltimore' root certificate, and we will update the internal root ca list to swap 'Digicert' with 'Baltimore' for the next public release.  Thanks for bringing this to our attention!

https://github.com/adafruit/Adafruit_WICED_Arduino/tree/development/examples/TLS/HttpsLargeData

I'll close this issue once the internal list is updated on development.

microbuilder, 2016-04-19:
This should be fixed in 0.5.1 if you update to the latest code and reflash FeatherLib.  Thanks for bringing the certificate change on Amazon's part to our attention, though:

```
HTTPS Large Data Example

Bootloader  : 1.0.0
WICED SDK   : 3.5.2
FeatherLib  : 0.5.1
Arduino API : 0.5.1

Please wait while connecting to: 'HOMENETWORK' ... Connected!

SSID        : HOMENETWORK (-59 dBm)
Encryption  : WPA_MIXED
MAC Address : 44:39:C4:EB:B9:64
Local IP    : 192.168.1.44
Gateway     : 192.168.1.1
Subnet Mask : 255.255.255.0

Connecting to 'adafruit-download.s3.amazonaws.com' port 443 ... OK
Requesting '/text_100KB.txt' ... OK
Total bytes received (including headers):
 - 100421 bytes in 9.52 seconds
 - Speed ~ 10.55 KB/s
Total data count: 100000 CRC32: 2385174509

------------
DISCONNECTED
------------
```

closed 2016-04-19 by microbuilder