adafruit/Adafruit_WICED_Arduino #36: Add I2C/Wire .endTransmission(bool)

microbuilder, 2016-04-05:
Re this thread, the Maple I2C drivers seem to be missing the 'bool' override for `.endTransmission` in the Wire library: https://forums.adafruit.com/viewtopic.php?f=57&t=93496&p=470758#p470758

- https://github.com/adafruit/Adafruit_WICED_Arduino/blob/master/stm32/libraries/Wire/WireBase.h#L93

As described here this should have an alternative override to indicate if we want to send the STOP condition or not: https://www.arduino.cc/en/Reference/WireEndTransmission

hathach, 2016-04-05:
ah thanks, I didn't know about it, I will double check and fix it later.

hathach, 2016-04-05:
Added the function to fix build error. Added the multiple message with repeated START is easy with native mcu code, but it is kind of awkward with current Arduino WireBase wrapper. We will implement it later if this has issue with any sensor device.

hathach, 2016-04-06:
#37

commit e9471ba0 (2016-04-05): resolve #36
add endTransmission(bool) to fix build error.
It will always send stop bit since repeated Start with multiple I2C message is a bit awkward to support with current implementation.
commit fdbed69b (2016-04-07): update wire to support stopBit for endTransmission and requestFrom
resolve #36 #37
closed 2016-04-07 by hathach