adafruit/Adafruit_WICED_Arduino #37: I2C Still failing with MPL3115A2 sensor

hack-o-holic, 2016-04-05:
As a follow up to the previous issue I was asked to open an issue here and reference our thread in the Adafruit support forums.
[http://forums.adafruit.com/viewtopic.php?f=57&t=93496&p=470977#p470977](url)

microbuilder, 2016-04-05:
Thanks for filing this.  There are some options missing in the Maple based Wire library around the STOP condition that will need to be added since some I2C sensors have specific requirements around STOP in the message sequence.

hathach, 2016-04-06:
#36

hathach, 2016-04-07:
@hack-o-holic please try to pull development branch and test again with the MPL3115A2, unfortunately, I don't have the sensor with me to test with. Please let us know if that work for you or not.

microbuilder, 2016-04-07:
You can download the development branch using this link: https://github.com/adafruit/Adafruit_WICED_Arduino/archive/development.zip

Or you can see the changes here: https://github.com/adafruit/Adafruit_WICED_Arduino/commit/fdbed69b1eb8534312307f6c57e3758f7ec0a153

Unfortunately this is one of the rare sensors that requires the STOP condition in an usual place, and neither Thach nor me happen to have one on hand, but I'll order one to test with from the shop. Sorry for the trouble.

hack-o-holic, 2016-04-07:
Works beautifully now!  Thanks for the excellent work!

closed 2016-04-07 by microbuilder