adafruit/Adafruit_WICED_Arduino #38: Disconnect callback for MQTT and TCP don't imply that connected() method returns false

mattlevy, 2016-04-07:
I've noticed that even though I do receive callbacks to my handlers when disconnected, the connected() method still returns true. This means that my connection loop, which only attempts to connect if connected() is false, never executes. If I add an express call to disconnect() to my handler, I do attempt a reconnect.

I'm not sure which behavior you want, but it should be either changed so that no manual call to disconnect() is necessary, or documented.

Note: I have only experienced this with the AdafruitTCP and AdafruitMQTT classes. I have not tried any others, although I would assume they're consistent.

hathach, 2016-04-07:
thanks for reporting, it is a  bug, we will try to fix it asap.

hathach, 2016-04-08:
@mattlevy if possible, could you provide your sketch. We could fix that quicker and make sure it work for you as well.

mattlevy, 2016-04-08:
Here's my sketch as a text file. You'll need to fill in your own wifi network, etc. It should be clear where to do that.

You may notice that I have a global boolean "subscribed". That's because it's possible to subscribe multiple times to the same feed - I don't know if that's intended or not.

[Disconnect_bug_sketch.txt](https://github.com/adafruit/Adafruit_WICED_Arduino/files/210836/Disconnect_bug_sketch.txt)

hathach, 2016-04-11:
should be resolved on development branch, and get closed when merged to master

commit 0bd9879b (2016-04-11): resolve #38, run part of disconnect routine when disconnect_callback is fired.
closed 2016-04-19 by hathach