adafruit/Adafruit_WICED_Arduino #39: Wifi supports US region only

vk7hse, 2016-04-18:
WICED Feather currently only supports 2.4GHz wifi CH1 through CH11 

```
python feather_dfu.py info
Feather
ST32F205RGY
363231313533471900400026
44:39:C4:EB:B4:7C
1.0.0
3.5.2
0.5.0
0.5.0
Mar 11 2016
```

My understanding is that the Broadcom chip is currently locked to the US 2.4GHz band plan so to keep it within the US regulation.

(I did a quick check on the 2.4GHz wifi band and noticed that for the USA CH12 and CH13 are not permitted)

microbuilder, 2016-04-28:
Thanks for the information, but since this isn't a bug with the library itself, but rather a piece of information about WICED, I'm going to close this issue.

astra1963, 2016-10-24:
Is it really true that european and japanese channels are locked by hardware? I can´t see something like this on the layout and the WICED module is sold worldwide so there should be no limitations from this side.

microbuilder, 2016-10-24:
Channels 1 to 11 are perfectly fine in Europe as well, see https://en.wikipedia.org/wiki/List_of_WLAN_channels#Interference_concerns

astra1963, 2016-10-24:
I know but there are APs I don´t have access to the channel choice that work on ch11 and 12.  It would really be great to have the possibility to access them.

microbuilder, 2016-10-24:
We'll look into whether these can be enabled as a command in FeatherLib, or whether this would require a separate FeatherLib build. It's a SW not a HW limitation, but we'll need to explore how to handle this in a way that's easy to maintain.

vk7hse, 2016-10-24:
Noticed some new comments with regards to the upper channels,  Although I raised this back in March, it was more to just make it known and so that people didn't RMA the device thinking it's faulty.  Easiest way to resolve would be make mention in the learning/documentation for the WICED.  ;)

hathach, 2016-10-27:
We just push a new code in development branch that probably resolve this. You can get/set the WiFi country using these constant (a featherlib upgrade is required).

https://github.com/adafruit/Adafruit_WICED_Arduino/blob/development/cores/maple/libmaple/adafruit_constants.h#L285

Set command will save the country code to nvm, it will last through power lost or reset. Check out the new example here

https://github.com/adafruit/Adafruit_WICED_Arduino/blob/development/libraries/AdafruitWicedExamples/WiFi/WifiCountry/WifiCountry.ino

We haven't tested if the set command could change the country on-the-fly or it needs to go through a reset cycle. Could you mind helping us to confirm this when connecting to your AP.

Note: there is an issue with Korean Republic, although it is on different chip
https://community.cypress.com/message/24846#24846

vk7hse, 2016-10-27:
Due to it being almost 3am (don't ask why I'm watching Netflix!) I'll run tests later in the morning and report back.

vk7hse, 2016-10-28:
Initial test of setting the region to Australia came back stating that the changes had been made. I then changed my router to CH13 and let the WICED do an AP scan (the one in wifi/examples however my router wasn't seen so I set to CH12 and again no show, next was CH11 and it displayed as expected.  

More testing to be done, I don't have any equipment to be able to view debugging direct from the chip,  so I'll obviously need to find a way to get more verbose output the see what is happening with the setting once changed.  

More to come...

astra1963, 2016-10-29:
Sorry that it took me so long to test. 
Tested it for Germany and it does exactly what expected. Works fine with channels 12, 13 and all others.

Thank you guys

astra1963, 2016-10-29:
With WLAN_COUNTRY = WIFI_COUNTRY_AUSTRALIA it works fine with 12 and 13, too

microbuilder, 2016-10-29:
Thanks, we'll include this as part of the newt release (0.6.2).

closed 2016-10-29 by microbuilder

hathach, 2016-10-31:
@vk7hse could you get it work with further testing. @astra1963  already confirmed that it worked with AU country.

vk7hse, 2016-10-31:
I haven't as yet, but I was going to grab a clean copy of the firmware
update from github, as I don't think the 0.62 was truly at that stage, I
think I'd messed up my local copy, I should get a chance to in the next day
to confirm,  to be honest I don't really have an issue with the upper
channels not being available, as I've populated lower down!  However I'll
get back to you with my findings as soon as humanly possible! (read that as
when I'm not playing Mr Mum!)

vk7hse, 2016-10-31:
WiFi Country Example
Bootloader  : 1.0.0
WICED SDK   : 3.5.2
FeatherLib  : 0.6.2
Arduino API : 0.6.2
Current configured Country: US
Change country to one defined by WLAN_COUNTRY
Please wait while connecting to: 'vk7hse.org' ... Connected!
SSID        : vk7hse.org (-49 dBm)
Encryption  : WPA2_AES
MAC Address : 44:39:C4:EB:B4:7C
Local IP    : 192.168.178.108
Gateway     : 192.168.178.1
Subnet Mask : 255.255.255.0

So once I let this run the country was changed to AU

WiFi Country Example
Bootloader  : 1.0.0
WICED SDK   : 3.5.2
FeatherLib  : 0.6.2
Arduino API : 0.6.2
Current configured Country: AU
Please wait while connecting to: 'vk7hse.org' ... Connected!
SSID        : vk7hse.org (-46 dBm)
Encryption  : WPA2_AES
MAC Address : 44:39:C4:EB:B4:7C
Local IP    : 192.168.178.108
Gateway     : 192.168.178.1
Subnet Mask : 255.255.255.0

I setup a router on ch13 and used the ScanNetworks sketch and I wasn't able to receive it. Next I uploaded SoftAP Client RSSI Example set it to use ch13 and it failed to set the channel

SoftAP Client RSSI Example
Configuring SoftAP
Starting SoftAP
SDEP_CMD_SOFTAP_START failed, Error: BADCHAN (2020)
--- FEATHER HALTED ---

The details of my WICED are as follows...

Feather
ST32F205RGY
363231313533471900400026
44:39:C4:EB:B4:7C
1.0.0
3.5.2
0.6.2
0.6.2
Oct 27 2016

I've got both the BATADC and SPIFCS bridged although I don't imagine they would have any impact on this.

hathach, 2016-10-31:
thanks for the feedback, we will double check it again.

vk7hse, 2016-10-31:
Further to this I attempted setting to Germany just to see if that made any
difference, but nope! 😟

Unfortunately I don't have any debugging equipment that can interrogate to
get what's happening within the mcu, I'm saving to buy the SEGGER J-Link
EDU - JTAG/SWD Debugger as I'm wanting to learn more about the inner
workings of this kind of stuff! I've been dropping hints to the family for
Xmas! 😉

hathach, 2016-11-01:
@vk7hse thanks, we will double check later, got into middle of high speed ADC issue :D

astra1963, 2016-11-01:
Do you need any data from my working environment? Can I give some assistance?

hathach, 2016-11-04:
@astra1963  Sorry for late response, just stuck with other issues. Since it works well for you, I think it is fine. Since we are late on the release schedule, we will release as it is and come back later if others encounter the issue as @vk7hse .

vk7hse, 2016-11-04:
Sounds like a plan!

Just out of curiosity though, is it possible that because I've got both of
the optional jumper pads bridged that might be a problem? I don't think
that it is, but just wondering if it's worth removing them and re-rest?
Thoughts?

microbuilder, 2016-11-04:
The solder jumpers don't matter here ... one enables SPI flash and the other enables monitoring the LIPO battery voltage through a voltage divider.

commit bd7e30dc (2016-10-27): add example description, update featherlib

resolve #39