adafruit/Adafruit_WICED_Arduino #44: feather_dfu on Mac OS X 10.11.5

amutiso, 2016-06-04:
I am unable to get feather_dfu to build on Mac OS X 10.11.5.

I get the following error:

copying /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/MacOS.so -> build/exe.macosx-10.11-intel-2.7/MacOS.so
error: [Errno 1] Operation not permitted: 'build/exe.macosx-10.11-intel-2.7/MacOS.so'

Have not yet found a way to get around the issue. 

I am using the installed python v2.7.10.

Messed around a bit with 2.7 versions of python from MacPort but that has not help.

microbuilder, 2016-06-04:
You should be using the python script on OS X ... the executables are really only for Windows where there are far more issues that can go wrong with libusb, etc.

closed 2016-06-04 by microbuilder

amutiso, 2016-06-04:
Hello,

I am not using Windows. I am using Mac OS X 10.11. 

I still can't get past the feather_dfu build error.

amutiso, 2016-06-04:
Also see the following. If try using the feather_dfu script from the command line I get the following result

feather_dfu$ python feather_dfu.py info
Traceback (most recent call last):
  File "feather_dfu.py", line 85, in <module>
    cli()
  File "/Library/Python/2.7/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "feather_dfu.py", line 67, in info
    resp = sdep.syscommand(SDEP_CMD_INFO)
  File "/Users/..../Documents/Arduino/hardware/Adafruit_WICED_Arduino/tools/source/feather_dfu/sdep.py", line 140, in syscommand
    usbdev = usb.core.find(idVendor=USB_VID, idProduct=USB_PID, backend=backend)
  File "/Library/Python/2.7/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

microbuilder, 2016-06-05:
Did you following the OSX learning guide to install both click and libusb/pyusb?  It's not finding the libusb backend.

https://learn.adafruit.com/introducing-the-adafruit-wiced-feather-wifi/os-x-setup#install-python-tools

microbuilder, 2016-06-05:
This is really a support issue not a PR or bug fix ... please repost on our monitored support forums if this doesn't work after following the learning guide, thanks.