adafruit/Adafruit_WICED_Arduino #53: DAC example doesn't work

hathach, 2016-07-27:
https://forums.adafruit.com/viewtopic.php?f=57&t=100553

kelu124, 2016-07-27:
Thanks for this quick follow-up! If you have tips, I'd be glad to contribute.

hathach, 2016-07-28:
mcu's peripherals code is based on https://github.com/rogerclarkmelbourne/Arduino_STM32 , but it could be our modification's bug as well. Unfortunately, we are working on another project, please give us a bit of time to troubleshoot the issue.

PS: I would be helpful if you could provide the sketch that reproduce the issue, just to make sure we are working on the same piece of code.

hathach, 2016-08-01:
look like the  https://github.com/rogerclarkmelbourne/Arduino_STM32 dac function use PA4 for DAC, but those pins are not available in final wiced feather version ( I remembered it was when prototyping).  https://learn.adafruit.com/introducing-the-adafruit-wiced-feather-wifi/board-layout . We are looking at another pins.

hathach, 2016-08-01:
while waiting for us to look up the STM32F2 datasheet, maybe you want to try out the channel 2, since PA5 is available as SCK.

```
#include <libmaple/dac.h>

void setup() 
{
  dac_init(DAC, DAC_CH2);     // Start DAC2 (pin PA5)
}

void loop() 
{
  uint32_t i = 0;
  for ( i=0; i<4096; i++ )
  {
    dac_write_channel(DAC, DAC_CH2, i);
  }  
  for ( i=4095; i>0; i-- )
  {
    dac_write_channel(DAC, DAC_CH2, i);
  }  
}
```

kelu124, 2016-08-04:
You said you managed to plug a LED and see the waveform on the LED, did you manage to get the waveform on a oscillo? If so, can you confirm you were using the stock _.ino_ file?

hathach, 2016-08-05:
I plug the LED and saw it is at ~ 30-40% brightness, I guess the example sketch produce some kind of waveform, but isn't sure. 

Google "Arduino Sawtooth" lead to this page https://www.baldengineer.com/sawtooth-waveform-with-the-arduino-dues-built-in-dac.html

[image]

Unfortunately, I have no access to an oscillo just yet to confirm it. I will ask others at Adafruit to confirm it later.

kelu124, 2016-08-06:
Back here, have tried on another computer, and now seems to work ([snipped here](https://gist.github.com/kelu124/b435126bf9d0ab9c84fd6ce950bba05f)) gives what I expect.

I'll just be noting that DAC values below 128 are somehow tricky to work with, as they do not appear on the bitscope measuring.

hathach, 2016-08-10:
thanks for confirmation, at least it is not useless at all :) , we are new with STM32 chip as well.

hathach, 2016-08-18:
For DAC value below 128, maybe we should open an different issue. Please update the forum to let others know if it work for you now, thanks.

closed 2016-08-19 by hathach