adafruit/Adafruit_WICED_Arduino #58: High-speed ADC acquisition

kelu124, 2016-09-10:
Hey, me again,

I'm trying to see how fast the Feather WICED ADC can go. I've been trying with ` analogRead()`, and at first sight, I'm getting data at around 50ksps. Will be posting the raw data later on to support this.

Now, I've begun digging into the maple ADC and corresponding code, and I've put together a piece of [code to test acquisition speed](https://gist.github.com/kelu124/ee02b6f98d8997404783aa3043d40c1e). This code compiles nicely, but doesn't work (nothing on serial). I'm unsure of the reason behind this, and I guess some registers need to be finetuned.

For example, the enumeration for STM32F1 differs from STM32F2, as said by [leaflab](https://github.com/leaflabs/leaflabs-docs/blob/master/source/libmaple/overview.rst). However, using `ADC_SMPR_1_5` during compilation works, but not the actual `ADC_SMPR_3`.

Working on this at the moment, but would appreciate if you had some ideas on what I could do next, I'll be updating/documenting the issue in the same time.

Cheers,

Luc

hathach, 2016-09-12:
hi @kelu124 , thank you for filing the issue. We are currently working on the major release 0.6.0 which introduce softAP and HTTPServer as well as SPI Flash + Fatfs. Please give us a bit of time, we will try to see what we can do. However, we almost don't touch these peripheral drivers forked from https://github.com/rogerclarkmelbourne/Arduino_STM32 . Maybe you should check there in the meantime, since they are the author of these pieces of code.

kelu124, 2016-09-12:
Hi @hathach. Thanks for the message. I hope the release is going well! There's absolutely no emergency on my side, I'll be happy to try to solve this issue. 
The reference you gave seems great, I'll be digging into it.
Again, thanks, and cheers.

hathach, 2016-10-26:
related issue https://forums.adafruit.com/viewtopic.php?f=57&p=527280

kelu124, 2016-10-27:
Thanks! I'm reaching close to 2Msps with 2 ADCs out of 3. Do you think it'd be interesting to get a sketch in the board examples?

microbuilder, 2016-10-27:
We're looking into this right now as well, but if you have any code feel free to post is here as a reference.

kelu124, 2016-10-27:
Superb!
I've been working on a ADC to UDP streaming piece of code. I'm reaching [these 2Msps](https://github.com/kelu124/echomods/blob/master/croaker/feather_tests/2ADC2UDP.ino) so far. Will report on other progress.

hathach, 2016-10-28:
great, we are working on this ADC speed. A useful Application Note from ST

[en.CD00258017.pdf](https://github.com/adafruit/Adafruit_WICED_Arduino/files/557640/en.CD00258017.pdf)

kelu124, 2016-10-28:
This one is great! The reference manual](https://github.com/kelu124/echomods/blob/master/croaker/datasheets/en.CD00225773_STM32F2x5_RefManual.pdf) should help as well, there are 3 ADCs on the WICED, when, at a first glance, the application note covers STM32 with 2 ADCs.

hathach, 2016-10-28:
Triple interleaved mode can get up to 6 MSPS. Basically calling analogRead() in loop() is no good, it can only be 120 ksps due to the fact that loop() is one of the least priority thread in wiced's rtos system (others are usb, networking etc...). The highspeed ADC need to be callback/interrupt based, we are still working on a user friendly API for that.

hathach, 2016-10-28:
Reference Manual is one of the useful but worst docs to read :(

hathach, 2016-11-04:
There is more work to do, but we just add an new ADC class that supports highspeed up to 2 MSPS for single channel mode with FIFO DMA support. Though the real application probably couldn't keep up with the rate. You need an scope to measure the speed. 

WARNING: for fast sample rate the buffer size should be increased as well to prevent buffer overrun error.

[image]

kelu124, 2016-11-04:
I've been through the referenced commit, and coudn't find the triple interleaved, is it part of the work to be done?

kelu124, 2016-11-04:
I've been through the referenced commit, and coudn't find the triple interleaved (except in an enum), is it part of the work to be done?

microbuilder, 2016-11-04:
The current code is probably the extent of what we'll add to the codebase for the moment, but pull requests are always welcome. Hopefully this is enough to get anyone started with the ADC and working with interrupts, etc. Unfortunately, we don't always have the resources to cover every use case being a small team with a lot of products to maintain or release, and triple interleaved ADC is a fairly specialised request that would take time to thoroughly test and implement, etc. Thanks for your understanding.

closed 2016-11-04 by microbuilder

hathach, 2016-11-04:
just want to mention that ADC can also use as analog watchdog (hardware support), which will fire an interrupt callback when voltage on the analog pin is out of threshold.

https://github.com/adafruit/Adafruit_WICED_Arduino/blob/development/libraries/AdafruitWicedExamples/Hardware/ADC/AnalogWatchdog/AnalogWatchdog.ino

kelu124, 2016-11-04:
No issue, I'll resume the work too =) Already reached close to 2Msps, but with 2 ADCs. Will try to learn with your approach! Thanks a lot @hathach

hathach, 2016-11-04:
thanks, just file issue if you found any bugs.

jonjedlicka, 2016-11-06:
How can you setup interrupt for two ADCs?

hathach, 2016-11-06:
0.6.2 does not support dual ADCs just yet. Ideally it will fill the buffer one after another alternatively and invoke callback when fifo is full.

jonjedlicka, 2016-11-07:
IC, i looked through the source code and there was mechanism to pass in 2 pins into the begin function, but when i build, i get errors.

microbuilder, 2016-11-07:
It's just a placeholder, as mentioned earlier in this thread multiple (interleaved) ADC pins isn't supported yet, but we're happy to integrate any pull requests. As it is, the 2 MSPS ADC on a single pin is already more than 99.9% of customers are likely to be able to use.

bhclowers, 2017-09-19:
It appears as though these initial functions help demonstrate the ability to capture data quickly but I'm not seeing exactly how this translates into some usable function.  How exactly can one acquire data at high speed and then do something useful (e.g. save or transfer) the buffer without losing too much time?  

From the example shown (adcSpeedTest.ino) it would be great to see how the buffer could be transfered via wifi.  I'm looking to identify a way to capture data continuously at just over 20 kSamples/sec. 

Is there a ping-pong buffer approach to making this happen?