adafruit/Adafruit_WICED_Arduino #60: Add missing .pulseIn Function

microbuilder, 2016-10-03:
.pulseIn isn't present in Arduino_STM32 except with the F1 core ... support should be added to the repo to enable this function for WICED, re: https://forums.adafruit.com/viewtopic.php?f=57&t=104028&p=521671#p520736

microbuilder, 2016-10-03:
Cross reference: https://github.com/rogerclarkmelbourne/Arduino_STM32/search?utf8=%E2%9C%93&q=pulsein

hathach, 2016-10-03:
thanks, the microDelay() is also not accurate with WICED as well #46 . Need to address that to get this done correctly.

hathach, 2016-10-11:
code from Arduino_STM32 seems to be test code, not work quite we like. I need to pull out code from avr core and port it to wiced. It works pretty precise, I have tested it using pulse from another UNO.

commit 5a12ede8 (2016-10-11): fix #60 pulsein Function
closed 2016-10-11 by hathach