adafruit/Adafruit_WICED_Arduino #66: Name Conflict with errorno

Andrewiski, 2016-12-24:
Including any library the includes errorno.h results in an error on compile 

error: cannot convert 'err_t' to 'err_t (*)()

renaming the pulic function in adafruit_sdep.h to something other then errno

ie 
`err_t        errno2       (void ) { return _errno; }
`
updating any Prints to reflect the name change allows code to be compiled.
`
Serial.printf(", Error: %s (%d)\r\n", errstr(), errno2());
`

hathach, 2017-01-12:
thanks, we will change err_t to something else in the next release

closed 2017-02-09 by hathach

Andrewiski, 2017-02-09:
awesome thanks

bornmand, 2017-05-17:
Looks like the current release of WICED BSP via the Arduino IDE (1.8.2)  is only version 0.6.2. I am still seeing the same error when including the library ArduinoJson.h in my project.

microbuilder, 2017-05-18:
This is fixed in the 'development' branch (https://github.com/adafruit/Adafruit_WICED_Arduino/tree/development), but it hasn't been released yet. We'll try to get a new release out with the current bug fixes early June!

bornmand, 2017-05-23:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } Thanks. Can I asked you a quick wiced question? Is there a method for OTA updates on the wiced, specifically via MQTT?

Andrewiski, 2017-05-26:
Not that I know of but I have never looked for it. I am sure if you went through the effort it could be added to the bootloader in the firmware but suspect you would be limited to half the memory as would need space to download then reboot, then reprogram as flashing directly via a wireless link is asking for a bricked device.

Andy

commit 0d4b32df (2017-02-01): fix #66 rename errno to errnum