adafruit/Adafruit_WICED_Arduino #67: status codes

thurber, 2017-01-11:
Thanks for the great library; I'm enjoying tinkering with my WICED!

It would be neat if the AdafruitHTTPServer object was able to set the status code of the response within a dynamic callback handler in addition to (or instead of) generating the content. For instance in the example file_not_found_generator, one might hope to do something like the following:

```c++
...
void file_not_found_generator (const char* url, const char* query, httppage_request_t* http_request)
{
  (void) url;
  (void) query;
  (void) http_request;
  httpserver.sendStatus(HTTP_STATUS_NOT_FOUND);
}
...
```

Alternatively, one might desire access to some type of `httppage_response_t` that could be manipulated in more advanced ways.

hathach, 2017-01-12:
Thanks for posting the suggestion. It is kind of  hard to provide a complete solution for the HTTP. Indeed for advanced pages, user may want to pack additional data into the HTTP header. Currently a simple header is written with status = 200 before the generator is invoked. So that normal user only need to focus on the contents, which work for 90% of cases. 

For advanced control, maybe we will create another type of HTTPPage where the generator is responsible for sending the header as well (hence everything return to client). There is a bit learning curve for what to include in the header, but this way we will solve this type of request once and for all.

How does it sound ?? If you are not familiar HTTP Response Header, we will add some helper as well ( You've better be though).

PS: We are busy on a new exciting product. Please give us a bit of time, we will get you update when it is done.

thurber, 2017-01-12:
That makes a lot of sense to me. I'm pretty familiar with the headers so that won't be a problem. Thanks, and excited to learn about your new product!