adafruit/Adafruit_WICED_Arduino #69: C version of feather_dfu

microbuilder, 2017-01-26:
A binary/executable version of [feather_dfu](https://github.com/adafruit/Adafruit_WICED_Arduino/tree/master/tools/source/feather_dfu) needs to be produced to avoid some of the numerous issues with Python, specifically multiple instances of python and different versions of it.

microbuilder, 2017-01-26:
Cross reference (for example) https://forums.adafruit.com/viewtopic.php?f=57&t=104251&p=552850#p552850 ... though there are other examples, as well as installation issues with utility libraries like click being installed to the wrong version, etc.

A C version would avoid some of these problems.

hathach, 2017-02-02:
working on feather_dfu C tool, almost done, could reset/force dfu on both linux and windows, testing on OSX, but it should be easy enough. I will clean up and push update on the tool changes. Hopefully, this will solve this issue once and for all.

closed 2017-02-06 by hathach