adafruit/Adafruit_WICED_Arduino #75: Fractional seconds read from getISO8601Time() seems incorrect

JasonSickler, 2017-07-07:
I am looking at the ISO8601 time readout in <Adafruit_feather> from my Adafriut WICED device, after it has connected to my WiFi LAN (so the RTC is synced).  Using this code in loop():
`  iso8601_time_t t;`
`  Feather.getISO8601Time(&t);`
`  Serial.printf("\"%s\" = ",t.minute);`

I get the following series of readouts:
`"22:04.058176Z"`
`"22:04.063176Z"`
`"22:04.002640Z"`
`"22:04.007640Z"`
`"22:04.012640Z"`
`"22:04.017640Z"`
`"22:04.022640Z"`
`"22:04.027640Z"`
`"22:04.032640Z"`
`"22:04.037640Z"`
`"22:04.042640Z"`
`"22:04.047640Z"`
`"22:04.052640Z"`
`"22:04.057640Z"`
`"22:04.062640Z"`
`"22:04.002104Z"`
`"22:04.007104Z"`

The sub_seconds values appear to be wrong, counting up to around 0.06, then wrap around, and the first decimal place stays are zero.  This cycles around about 12 times per second.

FYI, Feather.getUtcTime() gives the correct epoch time in seconds.

I tried to dig into sdep(), which Feather.getISO8601Time() calls:
`return sdep(SDEP_CMD_GET_ISO8601_TIME, 0, NULL, NULL, iso8601_time);`
where iso8601_time is the pointer to a iso8601_time_t variable in which the result goes.

No luck so far.  It appears to be a bug, or I'm doing something wrong.  Thanks in advance for the help.

hathach, 2017-07-17:
The issue is fixed in the development branch, we are testing to release the 0.6.5 as soon as possible.

hathach, 2017-07-18:
0.6.5 is live, please update and check to see if it fixes the issue.

JasonSickler, 2017-07-19:
Ah, I see that the issue was identified already in March.

When I download the .zip release of 0.6.5, and try to install the contents in the Arduino IDE, I get that the library is invalid.  That's true whether I use "Include Library-->Add .ZIPLibrary", manually extract to \Arduino\libraries\ or manual extract to [PERSONAL FOLDER\libraries\.  IDE Version is 1.8.2.

Perhaps I am doing something wrong?

hathach, 2017-07-19:
Hmm, can you remove the current BSP first and then install 0.6.5 using Arduino IDE -> Board Manager which is easier than downloading the zip package yourself.

JasonSickler, 2017-07-19:
I started down the road of removing old version first, but then went right to manually putting in the new version.  I needed to manually remove old version, and then go back to the library manager, instead of installing new version manually.

Compiles and uploads now.  Unfortunately, from what I can tell, the same problem exists in the timestamping as I described above.

hathach, 2017-07-19:
Did you upgrade the featherlib which contains the fix for the issue, choose the Section Featherlib instead of User Code and click upload icon

JasonSickler, 2017-07-19:
The entire WICED BSP installed through the Library Manager.  I manually went in to the folders to see that \featherlib\featherlib.* files were both the new versions, and they appear to be (2017-07-17 6:28AM modified date, sizes match up with GitHub).

hathach, 2017-07-19:
You will need to flash featherlib on the wiced board. Please choose
Tool -> Section -> Feather Lib
then click upload Icon to upgrade the featherlib

https://learn.adafruit.com/introducing-the-adafruit-wiced-feather-wifi/arduino-ide-setup#setting-the-section

JasonSickler, 2017-07-19:
Ah, good.  Just my ignorance.  Vaguely recall doing this back when I started, but forgot.  

And looks good!  You guys are awesome, thank you!

hathach, 2017-07-19:
No problems at all, we will close this issue now.

closed 2017-07-19 by hathach