adafruit/Adafruit_WICED_Arduino #85: WICED i2c and Adafruit 2.4" TFT

tpitman, 2018-11-15:
I just bought the WICED and have a i2c sensor connected to it. I also want to run the Adafruit 2.4" tft display.

I have the latest of all libraries installed as far as I can tell. Just downloaded everything yesterday.

The i2c sensor works fine all by itself. The tft works fine all by itself.

If I have both connected and trying to drive them then the tft shows what it needs to, but the sensor doesn't work at all.

If I have the tft connected, but the tft code commented out, then the sensor will run for a few seconds and then I stop getting values as if the tft hardware does something after a few seconds.

The pinouts for the tft don't show that they use the i2c lines at all, so I am confused.

Please advise how to fix / diagnose this issue.

ladyada, 2018-11-15:
hiya for tech support questions like this - please post in the forums - even if its a WICED/STM32 core issue, its easier for us to diagnose there, we'll open an specific issue if its warranted!

closed 2018-11-15 by ladyada

tpitman, 2018-11-15:
Ok, thanks you. I had already posted in the forums. Hopefully an answer will come soon? Wasn't sure whether this would be a better avenue.